kauailabs/allwpilib #8: Error updating firmware

cdewane, 2019-07-16:
Error updating firmware:
System.DllNotFoundException: Unable to load DLL 'STDFUFiles.DLL': The application has failed to start because its side-by-side configuration is incorrect. Please see the application event log or use the command-line sxstrace.exe tool for more detail. (Exception from HRESULT: 0x800736B1)

Google "stdfufiles side by side" leads to this which reccomends installing "Microsoft Visual C++ 2005 Redistributable Package (x86)":
https://www.chiefdelphi.com/t/nav-x-not-working/148486/3

This fixed the problem. Successfully upgraded firmware to 3.0.405

[image]

[image]

kauailabs, 2019-07-17:
Thanks for reporting this and the details.  v. 3.0.406 of the VMX-pi Tools for Windows installer has been released which modifies the installer to run the vcredist_x86 installer (for visual studio 2013) as part of the install.  This can be downloaded at:  https://www.kauailabs.com/public_files/vmx-pi/vmx-pi.zip

closed 2019-07-17 by kauailabs

amoose136, 2019-10-04:
Pretty sure this issue is not. Fixed. I did a fresh install and got the same error.

amoose136, 2019-10-04:
I needed `Microsoft Visual C++ 2005 Service Pack 1 Redistributable Package MFC Security Update` specifically the x86/32bit version _not_ the 64 bit version, then it updated without hitch.

kauailabs, 2019-10-05:
Thanks, @amoose136 , we've just released version 3.0.411 of the VMX-pi tools installer, it now includes and invokes the MSVC 2005 SP1 redist (x86).  This update also includes 3d design files for a Raspberry 4B-compatible VMX-pi enclosure.