kauailabs/allwpilib #9: Unable to build code. "Could not resolve all dependencies for configuration ':native_cameraserver'"

pit270, 2019-07-20:
I Am currently trying to configure FRC VS Code to build for the VMX-pi, I have followed and gone over the steps provided and have successfully set up gradle to use the custom gradleRIO repo by running the `gradlew publishToMavenLocal -PlocalPublish ` command, and am using the sample build.gradle file provided from the documentation. 

When building the code with the new build.gradle (the one for vmx), the code fails to build (more on that below), but the code still does build when using the original build.gradle (the one setup for roborio).

Here is the output when trying to build:
```
> Executing task: gradlew build  -Dorg.gradle.java.home="C:\Users\Public\frc2019\jdk" <

> Configure project :
INFO: Could not find files for the given pattern(s).

FAILURE: Build failed with an exception.

* What went wrong:
Could not determine the dependencies of task ':linkFrcUserProgramReleaseExecutable'.
> Could not resolve all dependencies for configuration ':native_cameraserver'.
   > Could not find edu.wpi.first.cameraserver:cameraserver-cpp:2019.4.1.
     Searched in the following locations:
       - file:/C:/Users/Public/frc2019/maven/edu/wpi/first/cameraserver/cameraserver-cpp/2019.4.1/cameraserver-cpp-2019.4.1.pom
       - file:/C:/Users/Public/frc2019/maven/edu/wpi/first/cameraserver/cameraserver-cpp/2019.4.1/cameraserver-cpp-2019.4.1-linuxathenadebug.zip
       - https://repo1.maven.org/maven2/edu/wpi/first/cameraserver/cameraserver-cpp/2019.4.1/cameraserver-cpp-2019.4.1.pom
       - https://repo1.maven.org/maven2/edu/wpi/first/cameraserver/cameraserver-cpp/2019.4.1/cameraserver-cpp-2019.4.1-linuxathenadebug.zip
       - http://devsite.ctr-electronics.com/maven/release/edu/wpi/first/cameraserver/cameraserver-cpp/2019.4.1/cameraserver-cpp-2019.4.1.pom
       - http://devsite.ctr-electronics.com/maven/release/edu/wpi/first/cameraserver/cameraserver-cpp/2019.4.1/cameraserver-cpp-2019.4.1-linuxathenadebug.zip
       - http://wpimirror.imjac.in/m2/release/edu/wpi/first/cameraserver/cameraserver-cpp/2019.4.1/cameraserver-cpp-2019.4.1.pom
       - http://wpimirror.imjac.in/m2/release/edu/wpi/first/cameraserver/cameraserver-cpp/2019.4.1/cameraserver-cpp-2019.4.1-linuxathenadebug.zip
     Required by:
         project :

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output. Run with --scan to get full insights.

* Get more help at https://help.gradle.org

BUILD FAILED in 3s
The terminal process terminated with exit code: 1
```

I have also uploaded a full gradle scan here: https://scans.gradle.com/s/het4y6ue723ug/failure?openFailures=WzBd#top=0

It's possible I simply have something configured wrong, I just can't figure out where.

kauailabs, 2019-07-20:
Thanks for posting the build error.  There's something that doesn't look right in the output:

cameraserver-cpp-2019.4.1-linuxathenadebug.zip

athena is the code name for RoboRIO, so it's using the wrong configuration.  When building for VMX-pi, we should be seeing something like:  "cameraserver-cpp-2019.4.1-linuxraspbian.zip".

I'm thinking it'd be good to see the build.gradle file in its entirety to figure out what might be causing this issue.  If you could post your entire project source code including the build.gradle, that would be ideal.

pit270, 2019-07-20:
It's worth noting that this error occurs even when using a default sample robot code base. I am using the provided template build.gradle. I have pasted it here for convenience.

```
plugins {
    id "cpp"
    id "google-test-test-suite"
    id "edu.wpi.first.GradleRIO" version "2019.4.1-VMX"
}

apply plugin: 'jaci.gradle.EmbeddedTools'

toolchainsPlugin.withRaspbian()

// Define my targets (RoboRIO) and artifacts (deployable files)
// This is added by GradleRIO's backing project EmbeddedTools.
deploy {
    targets {
        vmxpi("roborio") {
            // Team number is loaded either from the .wpilib/wpilib_preferences.json
            // or from command line. If not found an exception will be thrown.
            // You can use getTeamOrDefault(team) instead of getTeamNumber if you
            // want to store a team number in this file.
            team = frc.getTeamNumber()
        }
    }
    artifacts {
        frcNativeArtifact('frcCpp') {
            targets << "roborio"
            component = 'frcUserProgram'
            // Debug can be overridden by command line, for use with VSCode
            debug = frc.getDebugOrDefault(false)
        }
        // Built in artifact to deploy arbitrary files to the roboRIO.
        fileTreeArtifact('frcStaticFileDeploy') {
            // The directory below is the local directory to deploy
            files = fileTree(dir: 'src/main/deploy')
            // Deploy to RoboRIO target, into /home/lvuser/deploy
            targets << "roborio"
            directory = '/home/lvuser/deploy'
        }
    }
}

// Set this to true to include the src folder in the include directories passed
// to the compiler. Some eclipse project imports depend on this behavior.
// We recommend leaving this disabled if possible. Note for eclipse project
// imports this is enabled by default. For new projects, its disabled
def includeSrcInIncludeRoot = false

// Set this to true to enable desktop support.
def includeDesktopSupport = false

model {
    components {
        frcUserProgram(NativeExecutableSpec) {
            targetPlatform wpi.platforms.raspbian
            if (includeDesktopSupport) {
                targetPlatform wpi.platforms.desktop
            }

            sources.cpp {
                source {
                    srcDir 'src/main/cpp'
                    include '**/*.cpp', '**/*.cc'
                }
                exportedHeaders {
                    srcDir 'src/main/include'
                    if (includeSrcInIncludeRoot) {
                        srcDir 'src/main/cpp'
                    }
                }
            }

            // Defining my dependencies. In this case, WPILib (+ friends), and vendor libraries.
            useLibrary(it, "wpilib")
            wpi.deps.vendor.cpp(it)
        }
    }
    testSuites {
        frcUserProgramTest(GoogleTestTestSuiteSpec) {
            testing $.components.frcUserProgram

            sources.cpp {
                source {
                    srcDir 'src/test/cpp'
                    include '**/*.cpp'
                }
            }

            useLibrary(it, "wpilib", "googletest")
            wpi.deps.vendor.cpp(it)
        }
    }
}
```

kauailabs, 2019-07-20:
It appears that the links to the FRC GradleRIO maven repo were incorrect; however we didn't see the issue in our testing because we had cached copies of the necessary libraries on our development computers.  It appears that the libraries were not cached on your development computer, and then Gradle attempted to download theme using the repo list provided to it, things failed.

To address this, the custom version of GradleRIO for VMX-pi has been updated to correctly specify the  mirror locations; we've verified it on our side by complete deleting all cached gradle libraries, then reproducing your issue, then updating to the latest GradleRIO build that fixes the issue; this worked for us.  Can you please try it and let us know your results?

First, change to the directory containing the VMX-pi GradleRIO code, then:

- pull the latest code (e.g., using the "git pull" command)
- rebuild/publish the code to your local maven repo:
gradlew publishToMavenLocal -PlocalPublish

Thanks,

- scott

pit270, 2019-07-20:
This fixed the issue. Still running into some other problems but they are not related to this afaik.

Thanks!

kauailabs, 2019-07-20:
Closed based on feedback from @pit270.

closed 2019-07-20 by kauailabs