kauailabs/allwpilib #10: Issue with custom Phoenix-vmx library.

pit270, 2019-07-23:
I am having problems with the custom Phoenix-vmx library, as the build task fails as soon as it tries to create the library. Below is a paste of the console output:
```
> Executing task: gradlew build  -Dorg.gradle.java.home="C:\Users\Public\frc2019\jdk" <

> Configure project :
INFO: Could not find files for the given pattern(s).

FAILURE: Build failed with an exception.

* What went wrong:
A problem occurred configuring root project 'FRC2020'.
> Exception thrown while executing model rule: WPINativeJsonDepRules#addJsonLibraries(NativeDepsSpec, PlatformContainer, ExtensionContainer)
   > Cannot create 'libraries.ab676553-b602-441f-a38d-f1296eff6538CTRE_PhoenixPlatform_headers' using creation rule 'WPINativeJsonDepRules#addJsonLibraries(NativeDepsSpec, PlatformContainer, ExtensionContainer) > create(ab676553-b602-441f-a38d-f1296eff6538CTRE_PhoenixPlatform_headers)' as the rule 'WPINativeJsonDepRules#addJsonLibraries(NativeDepsSpec, PlatformContainer, ExtensionContainer) > create(ab676553-b602-441f-a38d-f1296eff6538CTRE_PhoenixPlatform_headers)' is already registered to create this model element.

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output. Run with --scan to get full insights.

* Get more help at https://help.gradle.org

BUILD FAILED in 3s
The terminal process terminated with exit code: 1
```

Also, here is a gradle scan of the build task: https://scans.gradle.com/s/pi7dpbv6fua6i/

Here is what I have done (using c++): Created a sample robot template and am using provided build.gradle file. I am using the latest version of the custom gradleRIO, installed the custom navx library (without issue), installed the phoenix-vmx library. I have also included the necessary `c_FeedEnable(100);` code, although the issue is occuring before the build process reaches the code.

If you need any more info let me know. Thank you!

kauailabs, 2019-07-23:
@pit270, it looks like the Phoenix-vmx.json file that was on our site was incorrect.   Specifically for C++, in this group:

        {
            "groupId": "com.kauailabs.vmx.first.hal",
            "artifactId": "hal-cpp",
            "version": "2019.4.1-vmxpi-beta-8",
            "libName": "CTRE_PhoenixPlatform",
            "sharedLibrary": true,
            "skipInvalidPlatforms": true,
            "binaryPlatforms": [
                "linuxraspbian"
            ]
        }    

There should not be this line:

            "headerClassifier": "headers",

We've updated the Phoenix-vmx.json on our site w/this fix.  So you can either edit your Phoenix-vmx.json file locally, or delete the Phoenix-vmx.json vendordeps file from your project, and then re-install it from:

http://www.kauailabs.com/dist/frc/2019/Phoenix-vmx.json

Let me know how that goes, I'll keep the issue open until I hear back from you.

pit270, 2019-07-24:
Now when I build, this is the output. Based on how it looks, it appears to be a similar problem to issue #9. 
```
> Executing task: gradlew build  -Dorg.gradle.java.home="C:\Users\Public\frc2019\jdk" <

> Configure project :
INFO: Could not find files for the given pattern(s).

FAILURE: Build failed with an exception.

* What went wrong:
Could not determine the dependencies of task ':linkFrcUserProgramReleaseExecutable'.
> Could not find canutils-linuxraspbian.zip (com.ctre.phoenix:canutils:5.14.1).
  Searched in the following locations:
      file:/C:/Users/Public/frc2019/maven/com/ctre/phoenix/canutils/5.14.1/canutils-5.14.1-linuxraspbian.zip

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output. Run with --scan to get full insights.

* Get more help at https://help.gradle.org

BUILD FAILED in 2s
Dependency Error!
GradleRIO detected this build failed due to missing dependencies!
Try again with `./gradlew downloadAll` whilst connected to the internet (not the robot!)
If the error persists, ensure you are not behind a firewall / proxy server (common in schools)
The terminal process terminated with exit code: 1
```

Tried running `./gradlew downloadAll` and received the same error. Also, it is worth noting that the failure will switch between :linkFrcUserProgramReleaseExecutable and :linkFrcUserProgramDebugExecutable, probably just depending on which task finished first.

kauailabs, 2019-07-24:
Something seems strange about this part:

  Searched in the following locations:
      file:/C:/Users/Public/frc2019/maven/com/ctre/phoenix/canutils/5.14.1/canutils-5.14.1-linuxraspbian.zip

Once Phoenix-vmx.json is included in your project's vendordeps directory, when compiling should be searching in the locations specified in the Phoenix-vms.json file, which are at line 7 and 8:

    "mavenUrls": [
        "http://devsite.ctr-electronics.com/maven/release/",
        "https://repo1.maven.org/maven2/"
    ],

Can you think of why in your case the build is only searching at C:/Users/Public/frc2019/maven, rather than also searching at the locations in the Phoenix-vmx.json file?

What I did to verify this works on my side was:

- Delete the following sub-directories from these parent directories:

  C:\Users\<username>\.gradle\caches\modules-2\files-2.1
  C:\Users\Public\frc2019\maven

The deleted sub-directories to clean things out were:

com.ctre.phoenix
com.kauailabs.navx.frc
com.kauailabs.vmx.first.hal

Next, I cleaned out the gradle build cache for my C++ project that uses phoenix and navx:

- In VSCode, WPI Library Extension ,run gradle command "clean"

After doing this, all local caches of phoenix, vmx and navx libraries are cleared out.

Finally, I made sure I was connected to the internet (required because no libraries are longer cached locally), and then I rebuilt the project in VSCode, deployed it to VMX-pi and it ran as expected.

I'm not sure if if this might help, but I'm not able to reproduce the problem here. 

If none of the above helps, I'd recommend creating a .zip file of your project directory and posting it   If you'd like, you can post it to this [link](https://my.pcloud.com/#page=puplink&code=Wg07ZCLWRtgGktEYnhEzBifT0Sb4Spm37).

pit270, 2019-07-24:
I cleared out all the caches as you described and now the code builds. Weird, but its fixed now. Thanks! I guess something to note if others have a similar issue.

closed 2019-07-24 by pit270