kauailabs/allwpilib #13: Can't read Joystick input from VMX-pi

PixelGaMERCaT, 2019-07-31:
My joysticks are plugged in properly, but I can't read any button nor axis input from any joystick or controller. I'm reading from the joystick in teleopPeriodic and printing the y-axis and a button. However, driver station says that the axis or button is not available, and to check if the joystick is plugged in.

kauailabs, 2019-07-31:
Hmm, we haven't seen any problem in this area.  We might need some help reproducing it.  On the Driver Station, on the "USB" tab, what is displayed in the "USB Order" control (which USB Device #, and what kind of device are displayed).  We have tested on several different USB Device #s without any trouble; is your joystick showing up as Device 0?

PixelGaMERCaT, 2019-07-31:
I've tested device numbers 4 and 5 (in code and driver station) and they haven't worked, but not 0. I'll test that next time I get a chance.

kauailabs, 2019-08-05:
This has been sorted out; previously, the input values from the highest numbered of the 6 possible joysticks was being treated as joystick 0.   This has been corrected (all 6 joysticks now work as expected) and the fix will be included in the next beta release build.

kauailabs, 2019-08-05:
Fixed in vmxpi-beta-12 release.

closed 2019-08-05 by kauailabs