kauailabs/allwpilib #14: Half-comms after deploy

PixelGaMERCaT, 2019-07-31:
After deploying code, driver station shows a half-green half-red bar for communications. This is fixed with a power cycle.

kauailabs, 2019-07-31:
Thanks; we've seen this too; the related symptom is that sometimes the console logging isn't working when this occurs.  If you experience trouble getting console logs, and until we get this resolved, you can directly view the logs on the raspberry by opening a console terminal and issuing these commands:

cd /var/local/kauailabs/log
tail -f FRCUserProgram.log

kauailabs, 2019-08-04:
The issue has been sorted out; the VMX-pi driver station code was not correctly reusing the server TCP/IP port; so if the robot app was shut down and restarted quickly, often it was not opening up a TCP port for the Driver Station to connect to.  When this occurred, the half-green/half-red communication bar would be displayed, and logging messages from the robot were not displayed at the driver station or RioLOG.  The logging area was reworked a bit too, to ensure that some log messages which were not being transmitted correctly to the driver station are now properly sent.  We'll be releasing this fix very soon in an upcoming release.

kauailabs, 2019-08-05:
Fixed in vmxpi-beta-12 release.

closed 2019-08-05 by kauailabs