kauailabs/allwpilib #15: Dependency error when building with CTRE-Phoenix-vmx

PixelGaMERCaT, 2019-08-06:
When I try to build with the CTRE-Phoenix-vmx library (http://www.kauailabs.com/dist/frc/2019/Phoenix-vmx.json), I get this error:
```
> Executing task: gradlew build   -Dorg.gradle.java.home="C:\Users\Public\frc2019\jdk" <

> Task :downloadDepsPreemptively FAILED

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':downloadDepsPreemptively'.
> Could not find cci-linuxraspbian.zip (com.ctre.phoenix:cci:5.14.1).
  Searched in the following locations:
      file:/C:/Users/Public/frc2019/maven/com/ctre/phoenix/cci/5.14.1/cci-5.14.1-linuxraspbian.zip

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output. Run with --scan to get full insights.

* Get more help at https://help.gradle.org

BUILD FAILED in 4s
2 actionable tasks: 1 executed, 1 up-to-date
Dependency Error!
GradleRIO detected this build failed due to missing dependencies!
Try again with `./gradlew downloadAll` whilst connected to the internet (not the robot!)
If the error persists, ensure you are not behind a firewall / proxy server (common in schools)
The terminal process terminated with exit code: 1
```

It builds fine without the library, and I don't have a cci-5.14.1-linuxraspbian.zip file. Am I missing something?

kauailabs, 2019-08-06:
Are you sure you have the latest KadenK version of GradleRIO installed on your machine, and are connected to the internet?  And have the Phoenix-vmx.json file in your project vendordeps?

The logs indicate that this build is only searching at this location:

  Searched in the following locations:
      file:/C:/Users/Public/frc2019/maven/com/ctre/phoenix/cci/5.14.1/cci-5.14.1-linuxraspbian.zip

Your Phoenix-vmx.json vendordeps file should contain this:

"mavenUrls": [
    "http://devsite.ctr-electronics.com/maven/release/",
    "https://repo1.maven.org/maven2/"
],

And so these sites should be searched for these files.

PixelGaMERCaT, 2019-08-06:
I've updated, and now it doesn't work.

PixelGaMERCaT, 2019-08-07:
I've found the files myself and downloaded them, and now it builds.

kauailabs, 2019-08-12:
To resolve this, we've got an updated Phoenix-vmx.json file.  This resolves an issue with Java and the phoenix libraries.  Please replace your existing CTRE Phoenix vendordeps file with the latest version (https://www.kauailabs.com/dist/frc/2019/Phoenix-vmx.json); this has been tested with Java and C++ and both are working in our testing.

closed 2019-08-12 by kauailabs