kauailabs/allwpilib #17: Some projects make the robot service automatically shut itself off

PixelGaMERCaT, 2019-08-07:
I have examples of working projects on my github. I'm not sure what causes this to happen.

kauailabs, 2019-08-12:
Can you please double check the projects on your github?  When I download this, I'm seeing two folders (NSP-PI, NSP-PI2) but they're empty.

Also, if logs are available that'd be great; it's not clear exactly how the symptoms (robot service automatically shut itself off) translate into.  

We have seen a problem when using Talons in Java that the robot app just freezes after initialization, the latest update to Phoenix-vmx.json resolves that, is it possible that this is what is being referred to?

PixelGaMERCaT, 2019-08-12:
I'm unsure. I'll double-check the projects once I get a chance.

kauailabs, 2019-09-01:
@PixelGaMERCaT, please retest this w/VMX-pi 2019.4.1 Beta Release 13.

kauailabs, 2019-09-17:
Similar to Issue#22, this scenario can be caused due to a conflict in signal handling with the java vm.  We will have a new Hal library available soon with the fix for this.

kauailabs, 2019-09-23:
This is fixed as of 2019.4.1 VMXpi Beta 16 Candidate (released 9/22/2019) and vmx-pi HAL version 1.1.207.

closed 2019-09-23 by kauailabs