kauailabs/allwpilib #29: Accessing Digital Output via wpilib throws exception.

cpeppler01, 2020-04-27:
The test involves using an external relay driven by a 5V Digital Output to power a 12V LED for lighting up vision.

This is what I did:

- Looked up the wpilib i/o mapping page on Kauai Lab beta website.
- Went down to section "Digital I/O (DIO) Channel Addressing"
- Selected wpilib channel 21, which should map to High Current DIO pin 10.
- Wired Ground, Power, and signal to the inputs on the relay (Relay shows a green LED, so it's happy).
- Assuming High Current DIO is set to 5V (as shipped), and direction set to output.

in 
in class Robot:
```
  private static final int ledChannelID = 21;
  private DigitalOutput ledOutput = new DigitalOutput(ledChannelID);
```

This the result (from Console):

```
VMX HAL:  pigpio library version 69 opened. ﻿
﻿﻿﻿﻿﻿﻿ VMX HAL:  SPI Aux Channel 2 opened with baudrate of 4000000. ﻿
﻿﻿﻿﻿﻿﻿ VMX HAL:  Established communication with VMX-pi model 0x32, hardware rev 60, firmware version 3.0.411 ﻿
﻿﻿﻿﻿﻿﻿ VMX HAL:  Acquired navX-Sensor configuration. ﻿
﻿﻿﻿﻿﻿﻿ VMX HAL:  Library version 1.1.220 ﻿
﻿﻿﻿﻿﻿﻿ Set VMX CAN Mode to NORMAL. ﻿
﻿﻿﻿﻿﻿﻿ Server Running... ﻿
﻿﻿﻿﻿﻿﻿ ********** Robot program starting ********** ﻿
﻿﻿﻿﻿﻿﻿ In disabledInit() ﻿
﻿﻿﻿﻿﻿﻿ Default disabledPeriodic() method... Override me! ﻿
﻿﻿﻿﻿﻿﻿ Default robotPeriodic() method... Override me! ﻿
﻿﻿﻿﻿﻿﻿﻿﻿ERROR ﻿﻿ 1 ﻿﻿ Unhandled exception: edu.wpi.first.hal.util.UncleanStatusException:  Code: -1098. HAL: A handle parameter was passed incorrectly ﻿﻿ frc.robot.Main.main(Main.java:27) ﻿﻿﻿
﻿﻿﻿﻿﻿﻿ Error at frc.robot.Main.main(Main.java:27): Unhandled exception: edu.wpi.first.hal.util.UncleanStatusException:  Code: -1098. HAL: A handle parameter was passed incorrectly ﻿
﻿﻿﻿﻿﻿﻿ 	at edu.wpi.first.hal.DIOJNI.getDIO(Native Method) ﻿
﻿﻿﻿﻿﻿﻿ 	at edu.wpi.first.wpilibj.DigitalOutput.get(DigitalOutput.java:69) ﻿
﻿﻿﻿﻿﻿﻿ 	at edu.wpi.first.wpilibj.smartdashboard.SendableBuilderImpl.lambda$addBooleanProperty$0(SendableBuilderImpl.java:219) ﻿
﻿﻿﻿﻿﻿﻿ 	at edu.wpi.first.wpilibj.smartdashboard.SendableBuilderImpl.updateTable(SendableBuilderImpl.java:95) ﻿
﻿﻿﻿﻿﻿﻿ 	at edu.wpi.first.wpilibj.livewindow.LiveWindow.updateValues(LiveWindow.java:290) ﻿
﻿﻿﻿﻿﻿﻿ 	at edu.wpi.first.wpilibj.IterativeRobotBase.loopFunc(IterativeRobotBase.java:263) ﻿
﻿﻿﻿﻿﻿﻿ 	at edu.wpi.first.wpilibj.TimedRobot.startCompetition(TimedRobot.java:81) ﻿
﻿﻿﻿﻿﻿﻿ 	at edu.wpi.first.wpilibj.RobotBase.startRobot(RobotBase.java:263) ﻿
﻿﻿﻿﻿﻿﻿ 	at frc.robot.Main.main(Main.java:27) ﻿
﻿﻿﻿﻿﻿﻿  ﻿
﻿﻿﻿﻿﻿﻿﻿﻿Warning ﻿﻿ 1 ﻿﻿ Robots should not quit, but yours did! ﻿﻿ edu.wpi.first.wpilibj.RobotBase.startRobot(RobotBase.java:274) ﻿﻿﻿
﻿﻿﻿﻿﻿﻿ ﻿Warning﻿ at edu.wpi.first.wpilibj.RobotBase.startRobot(RobotBase.java:274): Robots should not quit, but yours did! ﻿
﻿﻿﻿﻿﻿﻿﻿﻿ERROR ﻿﻿ 1 ﻿﻿ The startCompetition() method (or methods called by it) should have handled the exception above. ﻿﻿ edu.wpi.first.wpilibj.RobotBase.startRobot(RobotBase.java:276) ﻿﻿﻿
﻿﻿﻿﻿﻿﻿ Error at edu.wpi.first.wpilibj.RobotBase.startRobot(RobotBase.java:276): The startCompetition() method (or methods called by it) should have handled the exception above. ﻿
﻿﻿﻿﻿﻿﻿ # ﻿
﻿﻿﻿﻿﻿﻿ # A fatal error has been detected by the Java Runtime Environment: ﻿
﻿﻿﻿﻿﻿﻿ # ﻿
﻿﻿﻿﻿﻿﻿ #  SIGSEGV (0xb) at pc=0x739ccf28, pid=11649, tid=11698 ﻿
﻿﻿﻿﻿﻿﻿ # ﻿
﻿﻿﻿﻿﻿﻿ # JRE version: OpenJDK Runtime Environment (11.0.7+10) (build 11.0.7+10-post-Raspbian-3deb10u1) ﻿
﻿﻿﻿﻿﻿﻿ # Java VM: OpenJDK Server VM (11.0.7+10-post-Raspbian-3deb10u1, mixed mode, g1 gc, linux-) ﻿
﻿﻿﻿﻿﻿﻿ # Problematic frame: ﻿
﻿﻿﻿﻿﻿﻿ # C  [libvmxpi_hal_cpp.so+0xa3f28][thread 11704 also had an error] ﻿
﻿﻿﻿﻿﻿﻿ [thread 11725 also had an error] ﻿
﻿﻿﻿﻿﻿﻿   spiGoA+0x2a8 ﻿
﻿﻿﻿﻿﻿﻿ # ﻿
﻿﻿﻿﻿﻿﻿ # No core dump will be written. Core dumps have been disabled. To enable core dumping, try "ulimit -c unlimited" before starting Java again ﻿
﻿﻿﻿﻿﻿﻿ # ﻿
﻿﻿﻿﻿﻿﻿ # An error report file with more information is saved as: ﻿
﻿﻿﻿﻿﻿﻿ # //hs_err_pid11649.log ﻿
﻿﻿﻿﻿﻿﻿﻿﻿Warning ﻿﻿ 44004 ﻿﻿ FRC:  The Driver Station has lost communication with the robot. ﻿﻿ Driver Station ﻿﻿﻿

```

If I comment out the reference to DigitalOutput, all is happy.

(Just a note:  I just did a Raspi update/upgrade, and it looks like a new java runtime came down...not sure if that might have an effect.)

kauailabs, 2020-04-28:
Thanks for reporting this is indeed a bug.

Root Cause Analysis:  The VMX-pi HAL for WPI Library Programming fails to allocate DIO Channel Number (WPI Library Channel 21) due to a range-check error in DigitalInternal.cpp.   Because the allocation fails, the underlying handle that is returned from the allocation is an "invalid handle" value.  When later the "Get" (or any other) method is invoked using that handle, an error is returned.  In java, this is handled as an "unclean status" exception, which if unhandled (by default, it is unhandled) will cause the java robot application to abort.  This issue only occurs for Digital IO's; the same VMX-pi channel can be used successful for PWM.

To resolve this issue, an updated version of the VMX-pi WPI HAL Library will be required; this will also require an updated KauaiLabs GradleRIO release.

Workaround:  Use a different WPI Library Channel # (not 21) for Digital IO instead.

cpeppler01, 2020-04-28:
Per Scott's recommendation, I tested with channels 18-20, and they all worked correctly, toggling the relay, whereas channel 21 throws the exception.

kauailabs, 2020-05-18:
This issue is now closed; it's resolved in the 2020 version of the VMX-pi WPI Libraries. 

This issue is only valid now for the now-defunct 2019 version of the VMX-pi WPI Libraries.

closed 2020-05-18 by kauailabs