kauailabs/allwpilib #32: GroovyCompile exception on gradlew install of custom GradleRIO

cpeppler01, 2020-05-03:
After the announcement that VMXpi specific WPIlib 2020 was available for SparkMax, I started following the instructions on [this page](https://pdocs.kauailabs.com/vmx-pi/advanced/vmx-pi-for-frc-2020-robot-programming/).

The install of the 2020 version of WPIlib on the dev computer went through fine.  
The git clone of the custom GradleRIO went fine.

```
E:\FRC\Dev\GitHub>git clone https://github.com/kauailabs/GradleRIO.git
Cloning into 'GradleRIO'...
remote: Enumerating objects: 247, done.
remote: Counting objects: 100% (247/247), done.
remote: Compressing objects: 100% (103/103), done.
remote: Total 7291 (delta 94), reused 220 (delta 80), pack-reused 7044
Receiving objects: 100% (7291/7291), 13.57 MiB | 10.31 MiB/s, done.
Resolving deltas: 100% (3365/3365), done.

```
I hit a snag with the gradle install of the custom GradleRIO.  Here is the output I received:

```
E:\FRC\Dev\GitHub>cd GradleRIO

E:\FRC\Dev\GitHub\GradleRIO>dir
 Volume in drive E is DATA
 Volume Serial Number is 7020-04AB

 Directory of E:\FRC\Dev\GitHub\GradleRIO

05/03/2020  10:37 AM    <DIR>          .
05/03/2020  10:37 AM    <DIR>          ..
05/03/2020  10:37 AM               276 .gitignore
05/03/2020  10:37 AM               305 .travis.yml
05/03/2020  10:37 AM             1,389 azure-pipelines.yml
05/03/2020  10:37 AM             2,872 build.gradle
05/03/2020  10:37 AM    <DIR>          examples
05/03/2020  10:37 AM    <DIR>          gradle
05/03/2020  10:37 AM             5,947 gradlew
05/03/2020  10:37 AM             2,942 gradlew.bat
05/03/2020  10:37 AM    <DIR>          img
05/03/2020  10:37 AM             1,082 LICENSE
05/03/2020  10:37 AM             7,597 README.md
05/03/2020  10:37 AM    <DIR>          src
05/03/2020  10:37 AM             2,265 versionupdates.gradle
05/03/2020  10:37 AM    <DIR>          vscode-examples
05/03/2020  10:37 AM             1,503 wsl_install_java.sh
05/03/2020  10:37 AM    <DIR>          _localtest
              10 File(s)         26,178 bytes
               8 Dir(s)  1,669,426,282,496 bytes free
			   
E:\FRC\Dev\GitHub\GradleRIO>gradlew publishToMavenLocal -PlocalPublish

> Task :compileGroovy
startup failed:
E:\FRC\Dev\GitHub\GradleRIO\src\main\groovy\edu\wpi\first\gradlerio\wpi\WPIPlugin.groovy: 81: [Static type checking] - No such property: vmxPlatformVersion for class: edu.wpi.first.gradlerio.wpi.WPIExtension
 @ line 81, column 43.
           mau_config.setVersion(project.ex
                                 ^

E:\FRC\Dev\GitHub\GradleRIO\src\main\groovy\edu\wpi\first\gradlerio\wpi\WPIPlugin.groovy: 81: [Static type checking] - Cannot find matching method edu.wpi.first.nativeutils.configs.DependencyConfig#setVersion(java.lang.Object). Please check if the declared type is correct and if the method exists.
 @ line 81, column 21.
                       mau_config.setVersion(project.extensions.getByType(WPIExtension).vmxPlatformVersion)
                       ^

2 errors

> Task :compileGroovy FAILED

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':compileGroovy'.
> Compilation failed; see the compiler error output for details.

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output. Run with --scan to get full insights.

* Get more help at https://help.gradle.org

BUILD FAILED in 5s
3 actionable tasks: 3 executed

E:\FRC\Dev\GitHub\GradleRIO>
```

Please advise.

kauailabs, 2020-05-03:
Thanks for reporting this, a check-in was missed and is now updated.

Please pull the latest code from the Kauailabs Github GradleRIO repo and rebuild it.

closed 2020-05-03 by kauailabs

cpeppler01, 2020-05-03:
Just to confirm, this did indeed fix the problem.  Thank you for the prompt response.