kauailabs/allwpilib #33: Exception somewhere between JRE and HAL

cpeppler01, 2020-05-05:
I just recently upgraded to WPIlib 2020, and on Sunday was able to bring up the application developed with 2019 on 2020.

Today I tried to start things up, and received an unexpected surprise. I followed normal procedure, establishing network connection between dev computer and VMXpi.  I could bring up VNC viewer, ping, etc.....but FRC driver station would not establish communications with the robot software on the VMXpi.

I went to the Diagnostics panel, and the Robot IP address was green, but the Robot Communications icon was red:
[image]

I then used VSCode to build the robot program (successful), and then deployed robot code, which was also successful, so dev computer was able to establish communications with the VMXpi on both a network level, and an os/SSH level.

I then looked at the results of the gradle output, and saw the script used on the VMXpi to start the robot code running.  The script was in /home/lvuser/robotCommand.

I sshd into the VMXpi, and had the following session:

```
login as: pi
pi@10.78.22.2's password:
Linux raspberrypi 5.4.35-v7l+ #1314 SMP Fri May 1 17:47:34 BST 2020 armv7l

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Tue May  5 14:43:22 2020

SSH is enabled and the default password for the 'pi' user has not been changed.
This is a security risk - please login as the 'pi' user and type 'passwd' to set                                                                                                                                                              a new password.

pi@raspberrypi:~ $ cd /home/lvuser
pi@raspberrypi:/home/lvuser $ ls
BringUpSparkOnCAN.jar  deploy               readme
DataMonitor.jar        MotorController.jar  robotCommand
pi@raspberrypi:/home/lvuser $ cat robotCommand
/usr/local/frc/JRE/bin/java -XX:+UseConcMarkSweepGC -Djava.library.path=/usr/loc                                                                                                                                                             al/frc/third-party/lib -Djava.lang.invoke.stringConcat=BC_SB   -jar "/home/lvuse                                                                                                                                                             r/BringUpSparkOnCAN.jar"
pi@raspberrypi:/home/lvuser $ ./robotCommand
OpenJDK Server VM warning: Option UseConcMarkSweepGC was deprecated in version 9                                                                                                                                                             .0 and will likely be removed in a future release.
2020-05-05 14:47:14 initCheckPermitted:
VMX HAL:  Error initializing pigpio library.
System Resources required by the pigpio library are not available.
VMX HAL:  Unable to establish communication with VMX-pi circuit board via Raspbe                                                                                                                                                             rry PI Aux SPI Channel.
+---------------------------------------------------------+
|Sorry, you don't have permission to run this program.    |
|Try running as root, e.g. precede the command with sudo. |
+---------------------------------------------------------+

VMX HAL:  Error initializing VMX-pi HAL Library.
Failed to open default CAN Receive Stream.  CAN Initialization Failed!

MAU HAL:  Error expiring IO Watchdog.
MAU HAL:  Error expiring IO Watchdog.
MAU HAL:  Error expiring IO Watchdog.
Server Running...
Error Initializing VMX-pi HAL.  Robot app failed to start.
Error Initializing VMX-pi HAL.  Robot app failed to start.
Error Initializing VMX-pi HAL.  Robot app failed to start.
Error Initializing VMX-pi HAL.  Robot app failed to start.
Error Initializing VMX-pi HAL.  Robot app failed to start.
Error Initializing VMX-pi HAL.  Robot app failed to start.
^Cpi@raspberrypi:/home/lvuser $ ./robotCommand
OpenJDK Server VM warning: Option UseConcMarkSweepGC was deprecated in version 9                                                                                                                                                             .0 and will likely be removed in a future release.
2020-05-05 14:47:53 initCheckPermitted:
VMX HAL:  Error initializing pigpio library.
+---------------------------------------------------------+
System Resources required by the pigpio library are not available.
|Sorry, you don't have permission to run this program.    |
|Try running as root, e.g. precede the command with sudo. |
+---------------------------------------------------------+

VMX HAL:  Unable to establish communication with VMX-pi circuit board via Raspbe                                                                                                                                                             rry PI Aux SPI Channel.
[... 62 lines cut ...]
# No core dump will be written. Core dumps have been disabled. To enable core dumping, try "ulimit -c unlimited" before starting Java again
#
# If you would like to submit a bug report, please visit:
#   Unknown
#

---------------  S U M M A R Y ------------

Command Line: -XX:+UseConcMarkSweepGC -Djava.library.path=/usr/local/frc/third-party/lib -Djava.lang.invoke.stringConcat=BC_SB /home/lvuser/BringUpSparkOnCAN.jar

Host: rev 3 (v7l), 4 cores, 3G, Raspbian GNU/Linux 10 (buster)
Time: Tue May  5 14:48:52 2020 EDT elapsed time: 0 seconds (0d 0h 0m 0s)

---------------  T H R E A D  ---------------

Current thread is native thread

Stack: [0x72ea4000,0x736a3000],  sp=0x736a1a78,  free space=8182k
Native frames: (J=compiled Java code, A=aot compiled Java code, j=interpreted, Vv=VM code, C=native code)
C  [libvmxpi_hal_cpp.so+0xa3eb8]  spiGoA+0x238
C  [libvmxpi_hal_cpp.so+0xa4508]  spiGo+0x50
pi@raspberrypi:/home/lvuser $
```

Please advise.

kauailabs, 2020-05-05:
Permissions looks to be a culprit.  Have you changed the pi user password or anything like that?

cpeppler01, 2020-05-05:
pi user password is still the default.

kauailabs, 2020-05-05:
When’s the last time you’ve rebooted the robot?

cpeppler01, 2020-05-05:
Just recently.  I will do it again now and retry.

kauailabs, 2020-05-05:
Also please use the frcKillRobot.sh and frcRunRobot.sh commands to stop/start the robot app if using the command line.

cpeppler01, 2020-05-05:
Please note that I _did_ build OpenCV on this machine.  Would this create a conflict?

I did reboot, logged in immediately, and had this session:
```
login as: pi
pi@10.78.22.2's password:
Linux raspberrypi 5.4.35-v7l+ #1314 SMP Fri May 1 17:47:34 BST 2020 armv7l

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Tue May  5 16:15:56 2020

SSH is enabled and the default password for the 'pi' user has not been changed.
This is a security risk - please login as the 'pi' user and type 'passwd' to set                                                    a new password.

pi@raspberrypi:~ $ cd /home/lvuser
pi@raspberrypi:/home/lvuser $ sudo ./robotCommand
OpenJDK Server VM warning: Option UseConcMarkSweepGC was deprecated in version 9                                                   .0 and will likely be removed in a future release.
VMX HAL:  pigpio library version 69 opened.
VMX HAL:  SPI Aux Channel 2 opened with baudrate of 4000000.
Retrying Read....
VMX HAL:  Error reading from bank 0, address 0, length 4.
Read Aborted!
VMX HAL:  Established communication with VMX-pi model 0x32, hardware rev 60, fir                                                   mware version 3.0.411
VMX HAL:  Acquired navX-Sensor configuration.
VMX HAL:  Library version 1.1.220
Set VMX CAN Mode to NORMAL.
Server Running...
Exception in thread "main" java.lang.UnsatisfiedLinkError: 'boolean edu.wpi.firs                                                   t.hal.HAL.hasMain()'
        at edu.wpi.first.hal.HAL.hasMain(Native Method)
        at edu.wpi.first.wpilibj.RobotBase.startRobot(RobotBase.java:327)
        at frc.robot.Main.main(Main.java:27)
#
# A fatal error has been detected by the Java Runtime Environment:
#
[thread 1149 also had an error]#
SIGSEGV (0xb) at pc=0x75327e84, pid=1122, tid=1143
#
# JRE version: OpenJDK Runtime Environment (11.0.7+10) (build 11.0.7+10-post-Ras                                                   pbian-3deb10u1)
# Java VM: OpenJDK Server VM (11.0.7+10-post-Raspbian-3deb10u1, mixed mode, conc                                                   urrent mark sweep gc, linux-)
# Problematic frame:
# C  [libvmxpi_hal_cpp.so+0xa3e84]  spiGoA+0x204
#
# No core dump will be written. Core dumps have been disabled. To enable core du                                                   mping, try "ulimit -c unlimited" before starting Java again
#
# An error report file with more information is saved as:
# /home/lvuser/hs_err_pid1122.log
pi@raspberrypi:/home/lvuser $ cat /home/lvuser/hs_err_pid1122.log
#
# A fatal error has been detected by the Java Runtime Environment:
#
#  SIGSEGV (0xb) at pc=0x75327e84, pid=1122, tid=1143
#
# JRE version: OpenJDK Runtime Environment (11.0.7+10) (build 11.0.7+10-post-Raspbian-3deb10u1)
# Java VM: OpenJDK Server VM (11.0.7+10-post-Raspbian-3deb10u1, mixed mode, concurrent mark sweep gc, linux-)
# Problematic frame:
# C  [libvmxpi_hal_cpp.so+0xa3e84]  spiGoA+0x204
#
# No core dump will be written. Core dumps have been disabled. To enable core dumping, try "ulimit -c unlimited" before starting Java again
#
# If you would like to submit a bug report, please visit:
#   Unknown
#

---------------  S U M M A R Y ------------

[... 193 lines cut ...]
Dynamic libraries:
00010000-00011000 r-xp 00000000 b3:02 3994       /usr/lib/jvm/java-11-openjdk-armhf/bin/java
00020000-00021000 r--p 00000000 b3:02 3994       /usr/lib/jvm/java-11-openjdk-armhf/bin/java
00021000-00022000 rw-p 00001000 b3:02 3994       /usr/lib/jvm/java-11-openjdk-armhf/bin/java
01a08000-01b27000 rw-p 00000000 00:00 0          [heap]
6eef8000-6eef9000 ---p 00000000 00:00 0
6eef9000-6f6f9000 rw-p 00000000 00:00 0
6f6f9000-6f73a000 r-xp 00000000 b3:02 392896     /usr/local/frc/third-party/lib/libopencv_videostab.so.3.4
6f73a000-6f74a000 ---p 00041000 b3:02 392896     /usr/local/frc/third-party/lib/libopencv_videostab.so.3.4
6f74a000-6f74b000 r--p 00041000 b3:02 392896     /usr/local/frc/third-party/lib/libopencv_videostab.so.3.4
6f74b000-6f74c000 rw-p 00042000 b3:02 392896     /usr/local/frc/third-party/lib/libopencv_videostab.so.3.4
6f74c000-6f777000 r-xp 00000000 b3:02 392895     /usr/local/frc/third-party/lib/libopencv_videoio.so.3.4
6f777000-6f786000 ---p 0002b000 b3:02 392895     /usr/local/frc/third-party/lib/libopencv_videoio.so.3.4
6f786000-6f787000 r--p 0002a000 b3:02 392895     /usr/local/frc/third-party/lib/libopencv_videoio.so.3.4
6f787000-6f788000 rw-p 0002b000 b3:02 392895     /usr/local/frc/third-party/lib/libopencv_videoio.so.3.4
6f788000-6f78a000 rw-p 00000000 00:00 0
6f78a000-6f7c8000 r-xp 00000000 b3:02 392894     /usr/local/frc/third-party/lib/libopencv_video.so.3.4
6f7c8000-6f7d7000 ---p 0003e000 b3:02 392894     /usr/local/frc/third-party/lib/libopencv_video.so.3.4
6f7d7000-6f7d8000 r--p 0003d000 b3:02 392894     /usr/local/frc/third-party/lib/libopencv_video.so.3.4
6f7d8000-6f7d9000 rw-p 0003e000 b3:02 392894     /usr/local/frc/third-party/lib/libopencv_video.so.3.4
6f7d9000-6f7ef000 r-xp 00000000 b3:02 392892     /usr/local/frc/third-party/lib/libopencv_superres.so.3.4
6f7ef000-6f7ff000 ---p 00016000 b3:02 392892     /usr/local/frc/pi@raspberrypi:/home/lvuser $
```

I will run the same test, using frcKillRobot.sh and frcRunRobot.sh.  BRB

cpeppler01, 2020-05-05:
I rebooted, cd'd to /usr/local/vmxpi-frc/vmx_robot_manager , and ran  ./frcRunRobot.sh

Nada.  Did a "top".  Normally java will show up as major CPU process.  Not even there.

kauailabs, 2020-05-05:
Some native library the wpi library wants to use is not loading.  Perhaps it is opencv.  Opencv is included with wpi library.  I’d recommend starting with a fresh image.

The only other option i can think of is to build a simple c++ app - it links differently to the libraries than java at runtime and can maybe give more info on what native library won’t load.

cpeppler01, 2020-05-05:
OK, fresh image.  Is there one I can download and start over (and not install OpenCV)?

kauailabs, 2020-05-05:
Here's a link:  https://my.pcloud.com/publink/show?code=XZTAdLkZ9W8aVMP4orLcThV9m57zaXpI7lOV

cpeppler01, 2020-05-05:
Standard Etcher image creation I assume?

kauailabs, 2020-05-05:
https://pdocs.kauailabs.com/vmx-pi/advanced/vmx-pi-wpi-library-beta/beta-documentation/preconfigured-vmx_frc-sd-card-image-creation/

cpeppler01, 2020-05-05:
Bad news.

What I completed:

- Downloaded fresh image
- Flashed it to the same microSD.
- sudo apt-get update/upgrade
- Did the initial configuration, connected to network etc.
- did setupWifiAP.sh
- able to ping, connect correctly to right IP address.
- Followed the same steps as before
- Got a similar result.

On dev computer, FRC station can see the IP address, but Robot Communications is down.
VSCode can deploy robotCode.

Run top on VMXpi....no java running.
Run frcRunRobot.sh
run top again, still now java running.
cd /home/lvuser
sudo ./robotCommand
Result:
```
OpenJDK Server VM warning: Option UseConcMarkSweepGC was deprecated in version 9.0 and will likely be removed in a future release.
java.io.IOException: wpiHaljni could not be loaded from path or an embedded resource.
        attempted to load for platform /linux/athena/
Last Load Error:
no wpiHaljni in java.library.path: [/usr/local/frc/third-party/lib]

        at edu.wpi.first.wpiutil.RuntimeLoader.loadLibrary(RuntimeLoader.java:84)
        at edu.wpi.first.hal.JNIWrapper.<clinit>(JNIWrapper.java:38)
        at edu.wpi.first.wpilibj.RobotBase.startRobot(RobotBase.java:316)
        at frc.robot.Main.main(Main.java:27)
```

This might be a different error, but I did this with a fresh image....
Please advise.

KadenK, 2020-05-05:
The `attempted to load for platform /linux/athena/` seems to be the reason why it's failing, something is causing the code to look for a RoboRIO build. With a fresh image installed, I'm suspecting there might be something wrong on the dev pc side. Have you confirmed that your project's build.gradle file has been adapted properly to target the VMX-Pi?

cpeppler01, 2020-05-06:
Per issue #32,  I just recently updated the GradleRIO, and at the time, it was working.

I will go through those steps again and retest, if that would be helpful.

KadenK, 2020-05-06:
I was just able to recreate your issue. The root of the problem is that the GradleRIO install docs you are following were not fully updated to the 2020 version. On the [webpage linked](https://pdocs.kauailabs.com/vmx-pi/advanced/vmx-pi-for-frc-2020-robot-programming/) in #32, there are three screenshots presenting the three total modifications to the java build.gradle file. The third is the outdated one. In the 2020 update, the `dependencies{}` block has been expanded upon slightly. 

The 2020 build.gradle for a java project should have the following `dependencies{}` block:

```
dependencies {
    implementation wpi.deps.wpilib()
    nativeZip wpi.deps.wpilibJni(wpi.platforms.raspbian)
    nativeDesktopZip wpi.deps.wpilibJni(wpi.platforms.desktop)

    implementation wpi.deps.vendor.java()
    nativeZip wpi.deps.vendor.jni(wpi.platforms.raspbian)
    nativeDesktopZip wpi.deps.vendor.jni(wpi.platforms.desktop)

    testImplementation 'junit:junit:4.12'

    // Enable simulation gui support. Must check the box in vscode to enable support
    // upon debugging
    simulation wpi.deps.sim.gui(wpi.platforms.desktop, false)
}
```

From what I can tell, your current build.gradle is still using the RoboRIO platform. I believe your current problematic line is `nativeZip wpi.deps.wpilibJni(wpi.platforms.roborio)`

cpeppler01, 2020-05-06:
I have since reinstalled all the WPIlib 2020 software from scratch, and
re-executed the custom
GradleRIO bit.

This is the content of the existing dependencies block:

dependencies {
    compile wpi.deps.wpilib()
    compile wpi.deps.vendor.java()
    nativeZip wpi.deps.vendor.jni(wpi.platforms.raspbian)
    nativeDesktopZip wpi.deps.vendor.jni(wpi.platforms.desktop)
    testCompile 'junit:junit:4.12'
}

Note that the nativeZip does say raspbian.

I will replace the entire dependencies block with the one here in your
email, and let you know how it goes.

cpeppler01, 2020-05-06:
Updated the gradle file, with dependency block as you described (direct cut
and paste) redeployed.

Still get an error:

sudo ./robotCommand
OpenJDK Server VM warning: Option UseConcMarkSweepGC was deprecated in
version 9.0 and will likely be removed in a future release.
Exception in thread "main" java.lang.NoClassDefFoundError:
edu/wpi/first/wpilibj/TimedRobot
        at java.base/java.lang.ClassLoader.defineClass1(Native Method)
        at
java.base/java.lang.ClassLoader.defineClass(ClassLoader.java:1017)
        at
java.base/java.security.SecureClassLoader.defineClass(SecureClassLoader.java:174)
        at
java.base/jdk.internal.loader.BuiltinClassLoader.defineClass(BuiltinClassLoader.java:800)
        at
java.base/jdk.internal.loader.BuiltinClassLoader.findClassOnClassPathOrNull(BuiltinClassLoader.java:698)
        at
java.base/jdk.internal.loader.BuiltinClassLoader.loadClassOrNull(BuiltinClassLoader.java:621)
        at
java.base/jdk.internal.loader.BuiltinClassLoader.loadClass(BuiltinClassLoader.java:579)
        at
java.base/jdk.internal.loader.ClassLoaders$AppClassLoader.loadClass(ClassLoaders.java:178)
        at java.base/java.lang.ClassLoader.loadClass(ClassLoader.java:522)
        at frc.robot.Main.main(Main.java:27)
Caused by: java.lang.ClassNotFoundException:
edu.wpi.first.wpilibj.TimedRobot
        at
java.base/jdk.internal.loader.BuiltinClassLoader.loadClass(BuiltinClassLoader.java:581)
        at
java.base/jdk.internal.loader.ClassLoaders$AppClassLoader.loadClass(ClassLoaders.java:178)
        at java.base/java.lang.ClassLoader.loadClass(ClassLoader.java:522)

KadenK, 2020-05-06:
After a VSCode deploy, frcRunRobot.sh should be automatically called, and if you manually try to run robotCommand.sh it can cause issues. Can you redeploy your VSCode project and then check the log file located at `/var/local/kauailabs/logs/FRC_UserProgram.log`?

As a note, before running ./robotCommand or frcRunRobot.sh, it is best if you first run frcKillRobot.sh. Both frcRunRobot.sh and frcKillRobot.sh have been placed on the system PATH on the default VMX-Pi image, so they should both be runnable from anywhere in terminal.

cpeppler01, 2020-05-06:
Ran frcKillRobot.sh
Ran frcRunRobot.sh
Redeployed from VSCode

/var/local/kauailabs/logs/FRC_UserProgram.log:

cat /var/local/kauailabs/log/FRC_UserProgram.log
OpenJDK Server VM warning: Option UseConcMarkSweepGC was deprecated in
version 9.0 and will likely be removed in a future release.
Exception in thread "main" java.lang.NoClassDefFoundError:
edu/wpi/first/wpilibj/TimedRobot
        at java.base/java.lang.ClassLoader.defineClass1(Native Method)
        at
java.base/java.lang.ClassLoader.defineClass(ClassLoader.java:1017)
        at
java.base/java.security.SecureClassLoader.defineClass(SecureClassLoader.java:174)
        at
java.base/jdk.internal.loader.BuiltinClassLoader.defineClass(BuiltinClassLoader.java:800)
        at
java.base/jdk.internal.loader.BuiltinClassLoader.findClassOnClassPathOrNull(BuiltinClassLoader.java:698)
        at
java.base/jdk.internal.loader.BuiltinClassLoader.loadClassOrNull(BuiltinClassLoader.java:621)
        at
java.base/jdk.internal.loader.BuiltinClassLoader.loadClass(BuiltinClassLoader.java:579)
        at
java.base/jdk.internal.loader.ClassLoaders$AppClassLoader.loadClass(ClassLoaders.java:178)
        at java.base/java.lang.ClassLoader.loadClass(ClassLoader.java:522)
        at frc.robot.Main.main(Main.java:27)
Caused by: java.lang.ClassNotFoundException:
edu.wpi.first.wpilibj.TimedRobot
        at
java.base/jdk.internal.loader.BuiltinClassLoader.loadClass(BuiltinClassLoader.java:581)
        at
java.base/jdk.internal.loader.ClassLoaders$AppClassLoader.loadClass(ClassLoaders.java:178)
        at java.base/java.lang.ClassLoader.loadClass(ClassLoader.java:522)
        ... 10 more

kauailabs, 2020-05-06:
I think we are going to need to see your code the logs aren’t helping.  We need to see your vendor deps to know what other libraries are being linked in.

The other suggestion is to create a simple project, verify it works, then bring in the vendordeps one by one, making sure the 2020 version documented in the 2020 beta page are used.

KadenK, 2020-05-06:
I was just about to say the same. If you need some complete VSCode projects to test with, there are a few located here: https://github.com/KadenK/VMX-Pi-Test-Programs/tree/master/GradleRIO-FRCJ-2020

cpeppler01, 2020-05-06:
OK.

Pls find attached:

Robot.java
Main.java
build.gradle

I will start over with a new, fresh project.  No libraries, no
dependencies.  Bare bones.  We'll see if we can get that to work.

cpeppler01, 2020-05-06:
OK, starting over from scratch, with no dependencies was successful.

Build.gradle
```
plugins {
    id "java"
    id "edu.wpi.first.GradleRIO" version "2020.3.2-VMX"
}

sourceCompatibility = JavaVersion.VERSION_11
targetCompatibility = JavaVersion.VERSION_11

def ROBOT_MAIN_CLASS = "frc.robot.Main"

// Define my targets (RoboRIO) and artifacts (deployable files)
// This is added by GradleRIO's backing project EmbeddedTools.
deploy {
    targets {
        vmxpi("roborio") {
            // Team number is loaded either from the .wpilib/wpilib_preferences.json
            // or from command line. If not found an exception will be thrown.
            // You can use getTeamOrDefault(team) instead of getTeamNumber if you
            // want to store a team number in this file.
            team = frc.getTeamNumber()
        }
    }
    artifacts {
        frcJavaArtifact('frcJava') {
            targets << "roborio"
            // Debug can be overridden by command line, for use with VSCode
            debug = frc.getDebugOrDefault(false)
        }
        // Built in artifact to deploy arbitrary files to the roboRIO.
        fileTreeArtifact('frcStaticFileDeploy') {
            // The directory below is the local directory to deploy
            files = fileTree(dir: 'src/main/deploy')
            // Deploy to RoboRIO target, into /home/lvuser/deploy
            targets << "roborio"
            directory = '/home/lvuser/deploy'
        }
    }
}

// Set this to true to enable desktop support.
def includeDesktopSupport = false

// Defining my dependencies. In this case, WPILib (+ friends), and vendor libraries.
// Also defines JUnit 4.
dependencies {
    implementation wpi.deps.wpilib()
    nativeZip wpi.deps.wpilibJni(wpi.platforms.raspbian)
    nativeDesktopZip wpi.deps.wpilibJni(wpi.platforms.desktop)

    implementation wpi.deps.vendor.java()
    nativeZip wpi.deps.vendor.jni(wpi.platforms.raspbian)
    nativeDesktopZip wpi.deps.vendor.jni(wpi.platforms.desktop)

    testImplementation 'junit:junit:4.12'

    // Enable simulation gui support. Must check the box in vscode to enable support
    // upon debugging
    simulation wpi.deps.sim.gui(wpi.platforms.desktop, false)
}

// Setting up my Jar File. In this case, adding all libraries into the main jar ('fat jar')
// in order to make them all available at runtime. Also adding the manifest so WPILib
// knows where to look for our Robot Class.
jar {
    from { configurations.runtimeClasspath.collect { it.isDirectory() ? it : zipTree(it) } }
    manifest edu.wpi.first.gradlerio.GradleRIOPlugin.javaManifest(ROBOT_MAIN_CLASS)
}
```
Robot.java
```
/*----------------------------------------------------------------------------*/
/* Copyright (c) 2017-2018 FIRST. All Rights Reserved.                        */
/* Open Source Software - may be modified and shared by FRC teams. The code   */
/* must be accompanied by the FIRST BSD license file in the root directory of */
/* the project.                                                               */
/*----------------------------------------------------------------------------*/

package frc.robot;

import edu.wpi.first.wpilibj.Joystick;
// import edu.wpi.first.wpilibj.PWMVictorSPX;
import edu.wpi.first.wpilibj.TimedRobot;
// import edu.wpi.first.wpilibj.Timer;
// import edu.wpi.first.wpilibj.drive.DifferentialDrive;

/**
 * The VM is configured to automatically run this class, and to call the
 * functions corresponding to each mode, as described in the TimedRobot
 * documentation. If you change the name of this class or the package after
 * creating this project, you must also update the manifest file in the resource
 * directory.
 */
public class Robot extends TimedRobot {
  // private final DifferentialDrive m_robotDrive
  //     = new DifferentialDrive(new PWMVictorSPX(0), new PWMVictorSPX(1));
  private final Joystick m_stick = new Joystick(0);
//  private final Timer m_timer = new Timer();

  /**
   * This function is run when the robot is first started up and should be
   * used for any initialization code.
   */
  @Override
  public void robotInit() {
  }

  /**
   * This function is run once each time the robot enters autonomous mode.
   */
  @Override
  public void autonomousInit() {
    // m_timer.reset();
    // m_timer.start();
  }

  /**
   * This function is called periodically during autonomous.
   */
  @Override
  public void autonomousPeriodic() {
    // // Drive for 2 seconds
    // if (m_timer.get() < 2.0) {
    //   m_robotDrive.arcadeDrive(0.5, 0.0); // drive forwards half speed
    // } else {
    //   m_robotDrive.stopMotor(); // stop robot
    // }
  }

  /**
   * This function is called once each time the robot enters teleoperated mode.
   */
  @Override
  public void teleopInit() {
  }

  /**
   * This function is called periodically during teleoperated mode.
   */
  @Override
  public void teleopPeriodic() {
    // m_robotDrive.arcadeDrive(m_stick.getY(), m_stick.getX());
  }

  /**
   * This function is called periodically during test mode.
   */
  @Override
  public void testPeriodic() {
  }
}

```

So, should I add something back in, or have we found the source of the problem?

KadenK, 2020-05-06:
We aren't able to see your original project files, so we can't tell what the difference is. Could you zip your original project root folder and send it to kaden@kauailabs.com and scott@kauailabs.com? 
It seems like you previously tried to attach the Main.java, Robot.java, and build.gradle files of your project to your email which was converted into a GitHub comment, so the files were not posted. Additionally, those three files alone would not tell us what vendordeps you have installed, so the entire project, or at least all the vendordep .json's, would be preferable.

cpeppler01, 2020-05-06:
I deleted the build directory because it was massive, and temporary, but here is the 
entire directory zipped up, in case it helps to determine what went wrong.
[BringUpSparkOnCAN.zip](https://github.com/kauailabs/allwpilib/files/4584784/BringUpSparkOnCAN.zip)

cpeppler01, 2020-05-06:
I added the Spark Libraries back in, and have redeployed, and I can turn the motor, which is a good sign of life.

I have no idea what I did wrong (except for building OpenCV on the same raspi).  

You did find that doc bug for modifying the build.gradle file which is good.  I will keep adding stuff back in, and if I run into a problem I'll let you know.

KadenK, 2020-05-06:
Upon opening your project and trying it out straight out of the zip, I ran into the same error as you. I then swapped out the build.gradle for what I normally use on my projects, which can be found [here](https://github.com/KadenK/VMX-Pi-Test-Programs/blob/master/GradleRIO-FRCJ-2020/build.gradle). This worked perfectly fine.

I then compared the two versions of build.gradle, and I found the culprit. It is directly related to the change I previously had you make. 

The 2019 `dependencies{}` block is this:

dependencies {
     **compile** wpi.deps.wpilib()
     **compile** wpi.deps.vendor.java()
     nativeZip wpi.deps.vendor.jni(wpi.platforms.raspbian)
     nativeDesktopZip wpi.deps.vendor.jni(wpi.platforms.desktop)
     **testCompile** 'junit:junit:4.12'
 }

Note the bolded compile lines, and that there is a nativeZip line only for wpi.deps.vendor.java()

The 2020 `dependencies{}` block is this:

dependencies {
    **implementation** wpi.deps.wpilib()
    **nativeZip wpi.deps.wpilibJni(wpi.platforms.raspbian)**
    nativeDesktopZip wpi.deps.wpilibJni(wpi.platforms.desktop)

    **implementation** wpi.deps.vendor.java()
    nativeZip wpi.deps.vendor.jni(wpi.platforms.raspbian)
    nativeDesktopZip wpi.deps.vendor.jni(wpi.platforms.desktop)

    **testImplementation** 'junit:junit:4.12'

    // Enable simulation gui support. Must check the box in vscode to enable support
    // upon debugging
    simulation wpi.deps.sim.gui(wpi.platforms.desktop, false)
}

Note it has two nativeZip lines, this time one for wpi.deps.wpilibJni(). It also no longer uses "compile", but rather "implementation". This change in methods is once again used at the end of the build.gradle in the `jar{}` block

2019 `jar{}` block

jar {
    from { configurations.**compile**.collect { it.isDirectory() ? it : zipTree(it) } }
    manifest edu.wpi.first.gradlerio.GradleRIOPlugin.javaManifest(ROBOT_MAIN_CLASS)
}

2020 `jar{}` block

jar {
    from { configurations.**runtimeClasspath**.collect { it.isDirectory() ? it : zipTree(it) } }
    manifest edu.wpi.first.gradlerio.GradleRIOPlugin.javaManifest(ROBOT_MAIN_CLASS)
}

I'm not exactly sure at this point what these differences mean, but they do seem critical to a functioning 2020 robot. You can either replace both your `dependencies{}` and your `jar{}` blocks with what I've put here, or you can just copy the build.gradle that I have used that is in the above link. If you use the linked build.gradle file, it does currently contain a new GradleRIO installation/update method that is not yet documented, but functions much smoother and uses the same process as the official GradleRIO. Documentation should be out soon if you do choose to use it.

cpeppler01, 2020-05-06:
Glad you found the source of the problem.  Always bugs me if the root cause
is not found.

I think for now, I'm going to go to bed.  Thanks for the support!

closed 2020-05-13 by KadenK