kauailabs/allwpilib #37: Potential documentation missing or difficult to find?

MrRSquared, 2020-05-13:
Congratulations on the official FRC release, this is amazing software and even better hardware.  It is certainly opening doors for our team.  I am loving both the products and all of your excellent help.

I began trying to update our current bot using the new documentation and ran into a few bumps.

The first was when I wanted to update the software on the robot, I could not find the documentation for updating it at first. I eventually found it.
It is in...
VMX-pi for FRC Documentation>VMX_FRC Raspberry Pi Software Configuration>Manual VMX FRC SD Card Image Creation>Install/Update vmxpi-frc Tools Package

But people may want to know how to update the software on the robot without creating their own image, and that link was hidden for me (running Chrome) unless I clicked the link for the manual sd card creation.

The second issue I ran into which, if I was not doing anything wrong, is a bit more pressing is that I had a tough time getting the new gradle setup to build at first.

The steps on the website[website](https://pdocs.kauailabs.com/vmx-pi/software/vmx-pi-for-frc-2020-robot-programming/vmx-pi-for-frc-documentation/customizing-a-vscode-wpi-library-project-to-build-deploy-for-vmx-pi/)  imply that a user modifies the build.gradle file and then hits deploy.

However, I had to first visit [Kauai's Gradle Git](https://github.com/kauailabs/GradleRIO), clone the repo, install it with the directions on the Git 
`cd Gradle` 
`gradlew publishToMavenLocal -PlocalPublish`

Then, I had to build it, first disconnected from the robot, deploy.
After that, subsequent builds went smoothly.

Again, the issue could have been all with the user (me), but if not, there may be a few steps missing in the documentation.

KadenK, 2020-05-13:
Hi Mr. R^2,

As far as your first concern with [the location of updating vmxpi-frc](https://pdocs.kauailabs.com/vmx-pi/software/vmx-pi-for-frc-2020-robot-programming/vmx-pi-for-frc-documentation/vmx_frc-raspberry-pi-software-configuration/manual-vmx-frc-sd-card-image-creation/install-update-vmxpi-frc-tools-package/) goes, I can't give you much. I'll let Scott get back to you on that one.

Your second concern though is something I can help with. We are currently in a GradleRIO installation methodology switch, and not all the documentation has been fully updated yet. The steps on the website you linked to should be what is followed. I'll be updating the GradleRIO clone's README shortly to reflect the new changes. 

The methodology change I mentioned is meant to make it so that the user does not have to manually build the Kauailabs release of GradleRIO, which you say you had to do. A release of GradleRIO is now published and available on the [Gradle Plugin Portal](https://plugins.gradle.org/plugin/com.kauailabs.first.GradleRIO). To properly use this version, you must make sure that the GradleRIO plugin specified in your project's build.gradle file is the Kauailabs release. This means that the plugin{} block must contain the following line:

`id "com.kauailabs.first.GradleRIO" version "LATEST_VERSION"`
The latest version can be found on the [Kauailabs GradleRIO Gradle Plugin Portal page](https://plugins.gradle.org/plugin/com.kauailabs.first.GradleRIO) and is currently "2020.3.2-beta-2"

Previously the `id` property in that line was left at "edu.wpi.first.GradleRIO", and I think this might be why you were forced to build and locally publish the Kauailabs GradleRIO.

MrRSquared, 2020-05-13:
Thank you for your quick response, and that makes sense on both counts. 

It does seem I followed the directions correctly at first, and my grade build failed.
I used the following line.
`id "com.kauailabs.first.GradleRIO" version "2020.3.2-beta-1"`
when that failed, I followed the Gradle link and noticed you had already updated to beta 2 :)
and tried that too.
Unfortunately, I did not grab the error, but it was something to the effect of not being able to find the Gradle at com.kauailabs.first.GradleRIO.  It is possible I capitalized Kauai, but I thought after the first failure, I copied it from the Gradle site.
The only way it worked was to build it manually which was really not difficult merely a bit confusing for a spell.

KadenK, 2020-05-13:
If you are using the WPILib VSCode extension, then your issue might be that you are deploying your code. Whenever you run a "Deploy Robot Code" using the WPILib extension, it automatically deploys in offline mode. Because of this, GradleRIO only looks for a locally cached version of the Kauailabs GradleRIO. Try running a "Build Robot Code" instead. If this doesn't resolve your problem, then a copy of your build.gradle file would probably help speed things up a bit.

MrRSquared, 2020-05-13:
That sounds exactly like what happened.  I was unaware of this. Sorry for bothering you all.

KadenK, 2020-05-13:
No problem, we should probably include a note of this in the documentation.

KadenK, 2020-05-13:
Reopened because I realized that there were two parts to this issue

kauailabs, 2020-05-13:
The two sections of documentation related to this issue have been updated to (a) move the vmxpi-frc installation page to be more visible w/out requiring users to visit the "Manual SD Card Creation" page, and (b) indicate that whenever updating the Kauai Labs GradleRIO Plugin version, that the project should be rebuilt before redeploying.

This issue will stay open until we get verification that @MrRSquared that the issue is resolved.

MrRSquared, 2020-05-14:
Thank you all, this seems perfect.  I am going to close it; if I am overstepping, please reopen.

closed 2020-05-14 by MrRSquared